fr3ak2402/GalaxySlicer #15: Galaxy Slicer Profiles for Prusa MK3S

SaddamoDeBeers, 2023-07-23:
I am not sure if this is a bug or a feature. I did try to switch printer profiles in Galaxy Slicer, but it looks like not everything is working.
I do like Galaxy Slicer already, good job!

Profiles for K1 are working, Profiles for K1 Max are working, Voron profiles are working. I was trying switch to Prusa MK3S profiles, but these profiles do not show up, even I got the printers installed ( added ) I checked the directories and the profiles are "installed" after adding them, but they just dont work. There is a chance I am just blind.

[image]

and 

[image]

fr3ak2402, 2023-07-23:
Yes, unfortunately this is a bug, I have already fixed it and the fix will come in the next release.

If you want to temporarily fix the problem in your installed version of Slicer. You can replace the folder and file in the zip file with those in your installation folder.

The folder and file are in the resources/profiles folder.

[Prusa.zip](https://github.com/fr3ak2402/GalaxySlicer/files/12105539/Prusa.zip)

SaddamoDeBeers, 2023-07-23:
Thank you very much. I do like your slicer, good job.

I had to switch sometimes between PrusaSlicer and Superslicer, your Slicer combine them both together.

Have a great day.

closed 2023-07-24 by fr3ak2402