fr3ak2402/GalaxySlicer #31: K1 Max Not pausing

timmeade, 2023-08-05:
**Describe the bug**
K1 Max not pausing when pause at layer in gcode.    

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Created a simple slice with a pause at layer.
Exported gcode and uploaded to K1 Max.    It does not pause.

**Desktop (please complete the following information):**
Mac M1

fr3ak2402, 2023-08-07:
Replace the pause Gcode ,M400 U1 with PAUSE. This should fix the error.

timmeade, 2023-08-25:
It does.    Any chance this will be changeable somehow so we dont have to?

fr3ak2402, 2023-08-25:
So in the beta version

timmeade, 2023-08-25:
Thanks is that available now?

fr3ak2402, 2023-08-25:
Not yet, am still working on some things. 

If you want this right now, you can download the latest nightly build.... 

The only thing that is still shown in the nightly build is that the alpha2 version is shown as a new version because I was working on this window... you can just ignore this.

timmeade, 2023-08-25:
Much thanks.   Great work.

Tim

fr3ak2402, 2023-08-25:
The nightly version is fully functional except for this thing