fr3ak2402/GalaxySlicer #35: Something is not right with Travel calculations

kaushikbm, 2023-08-26:
Travel time in a print in GalaxySlicer is showing an estimate of 00:57:54 minutes where as in Creality Print is just 00:06:24 !! Following are the screenshots for comparison.

In GalaxySlicer
[image]

In Creality Print
[image]

fr3ak2402, 2023-09-27:
Is the problem still present in the beta version?

kaushikbm, 2023-09-28:
Yes. It still has the issue. I took a model. Kept the parameters in both the slicers more or less the same. GS-beta version is showing  42:58 51.9 % and CS is showing 5:18 10.8%

fr3ak2402, 2023-09-28:
Okay, is this just a display error or does it actually take longer to print?

kaushikbm, 2023-09-29:
It actually takes almost the same time it says.

kaushikbm, 2023-10-26:
@fr3ak2402 Could you find the issue ? I'm trying to setup the compiling environment in Windows. If it works maybe I'll also be able to contribute :)

kaushikbm, 2023-10-28:
As Orca slicer released 1.8.0 beta-1 today, I sliced one of my parts in it and was surprised to see the travel calculation was much better than before.

Then I started comparing profile parameters between OS and GS, and found that there were a few big differences in Galaxy slicer profiles. I started checking and updating profile parameters.

After updating them in GS results were at par with OS. And the TRAVEL calculations are **BETTER** in GS than OS !! 👯 :D

But still CS wins the game in travel calculations :D

In total print time OS is the least, than CS and then GS.

GS screenshot
[image]

OS screenshot
[image]

CS screenshot
[image]

fr3ak2402, 2023-10-28:
Thanks for the info and good job :)

I'll have a look at the parameters and touch up the profiles.

Then I'll see how I can improve the travel calculation. Or how that makes Creality / Cura.

closed 2024-02-04 by github-actions[bot]