fr3ak2402/GalaxySlicer #38: macOS GalaxySlicer.app is damaged and can't be opened

Rijswijker, 2023-09-04:
**Describe the bug**
After downloading GalaxySlicer I am not able to run the .app file.
macOS give an error "GalaxySlicer.app is damaged and can't be opened, you should move it to the Trash."
Redownloading make no sense, also downloading the first Alpha version results in the same error.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: maxOS 13.5.1
 - Type: Mac mini M2 Pro

Rijswijker, 2023-09-04:
Hadn't seen that you needed to run another command in the terminal ;-)
`sudo xattr -d com.apple.quarantine /Applications/GalaxySlicer.app`

closed 2023-09-04 by Rijswijker

fr3ak2402, 2023-09-04:
No problem, I was just about to reply.