fr3ak2402/GalaxySlicer #40: Preview for Creality K1

xpert13, 2023-09-09:
I have a Creality K1 printer and I noticed that generated by GalaxySlicer preview is not working in the printer GUI. 
I've made some research and found the problem:

GalaxySlicer generate preview like that:
```
; thumbnail begin 300x300 FILE_SIZE
; PREVIEW_FILE_CODE
; thumbnail end
```

But Creality K1 expect to get such code:
```
; png begin 300*300 FILE_SIZE
; PREVIEW_FILE_CODE
; thumbnail end
```

Could you please add ability to generate properly Creality K1 preview code?

fr3ak2402, 2023-09-10:
Yes of course I can do

bhast2, 2023-09-11:
i second this please

SHD3R, 2023-09-20:
There is no need for that. Setting G-code thumbnails to "96x96, 300x300" works fine for K1

xpert13, 2023-09-20:
> There is no need for that. Setting G-code thumbnails to "96x96, 300x300" works just fine for K1

Do you think it’s fine?
[image]

SHD3R, 2023-09-20:
> > There is no need for that. Setting G-code thumbnails to "96x96, 300x300" works just fine for K1
> 
> Do you think it’s fine?

Definitely not. Mine looks that way:
[image]

And here are my settings from OrcaSlicer.

[image]

fr3ak2402, 2023-09-20:
what firmware versions are you both using on your K1s?

xpert13, 2023-09-20:
My settings are similar:

[image]

And it's generate the same code as yours, but it's not working on my K1 (1.3.1.28).
I tried to generate gcode using Creality software, and it's generate a little bit different preview code (I've described it in the first post).

SHD3R, 2023-09-20:
Quite strange. I'm uploading generated file through K1 web interface. Maybe it adds something?

xpert13, 2023-09-20:
I’m using web interface too. I don’t know why your K1 works but it doesn’t meter: native Creality soft generates a bit different preview code and I just ask to give an opportunity to generate the same code in GS.

fr3ak2402, 2023-09-21:
I'm building this in so that it's created like Creality Print. Then this works in any case

fr3ak2402, 2023-09-29:
@xpert13 

I have a quick question:

```
; png begin 300*300 FILE_SIZE
; PREVIEW_FILE_CODE
; thumbnail end
```

instead of thumbnail end there must be png end?

```
; png begin 300*300 FILE_SIZE
; PREVIEW_FILE_CODE
; png end
```

xpert13, 2023-09-29:
@fr3ak2402 Yes, you are completely right. I've fixed my first post.

fr3ak2402, 2023-09-29:
@xpert13 
Okay, so now I'll put in the thumbnail thing for the K1 series. 

You will probably also have the problem that the AUX fan & Chamber fan of the K1s does not run. I have also solved this problem. So now you can also control the fans on the K1 & K1 Max like the BambuLab X1C etc.

fr3ak2402, 2023-10-05:
@xpert13 

I have created a gcode with the dev version, maybe you can try it on your K1.... and tell me if a thumbnail image is displayed on your printer?

[Thumbnail test.zip](https://github.com/fr3ak2402/GalaxySlicer/files/12818544/Thumbnail.test.zip)

fr3ak2402, 2023-10-05:
True. You're right, just fix that

xpert13, 2023-10-05:
I've fixed and it's work fine.

SHD3R, 2023-10-05:
@fr3ak2402 maybe look at the benchy that comes with the printer? I think the header looks different than header sent by @xpert13 and this is how CrealityPrint generates thumbnails for K1.
You can find this gcode here - https://github.com/CrealityOfficial/K1_Series_Klipper/blob/main/gcodes/K1/3DBenchy.gcode

fr3ak2402, 2023-10-05:
@xpert13 

Okay, I have adjusted this now, now it is indicated with * and no longer with x

[Cylinder_PLA_16m40s.zip](https://github.com/fr3ak2402/GalaxySlicer/files/12819634/Cylinder_PLA_16m40s.zip)

xpert13, 2023-10-05:
It's work fine. 
[image]

But there is no 96*96 preview inside the file

fr3ak2402, 2023-10-05:
@xpert13 

I had uploaded another zip which shows a cynlinder.

In this one I had also implemented the 96*96 thumbnail.

I have attached the zip again.
[Cylinder_PLA_16m40s.zip](https://github.com/fr3ak2402/GalaxySlicer/files/12820200/Cylinder_PLA_16m40s.zip)

SHD3R, 2023-10-05:
@fr3ak2402  Unfortunately, I can't do this today because I'm waiting for the new K1. I had to return the previous one because of early batch hardware problems.

fr3ak2402, 2023-10-10:
@SHD3R

Have you been able to test the Gcode on your K1?

closed 2023-10-11 by fr3ak2402

SHD3R, 2023-10-14:
> @SHD3R
> 
> Have you been able to test the Gcode on your K1?

works for me too