fr3ak2402/GalaxySlicer #44: Different model in gcode than what I sliced

martinoshub, 2023-09-23:
### GalaxySlicer Version

1.0.0-beta

### OS version

Windows 10

### OS architecture

x64

### Additional system information

First time user of Galaxy Slicer and I'm not sure if I'm messing up something or there's a bug. Apologies in advance if I'm messing up something. But even if I am, it means there's something unintuitive with the GUI. I've used 3 different slicer programs before.

The issue: I've tried 4 times to import an STL, slice it (adding a pause at a layer) and then save the gcode. Every time the end result (based on the gcode thumbnail) is a superhero kid figurine that I had not even seen on the screen let alone sliced it. I don't know why the software is generating gcode for a completely different model that I haven't even seen in model form.

### Printer

Creality K1

### How to reproduce

Open simple model (some 20x20x20mm cube with a couple of cylinders at the top before the top layer). Go to layer before holes are closed. Right click, pause. Click "slice plate" at the top. Change to "slice all". Click it. Export g-code.

### Actual results

Hero boy model gcode

### Expected results

The right model being sliced and gcoded

### Project file & Debug log uploads

[magnetcube.zip](https://github.com/fr3ak2402/GalaxySlicer/files/12705761/magnetcube.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

martinoshub, 2023-09-23:
I tried a print, anyway. It looks like the cube is being printed, but the thumbnail is of the hero kid. If the thumbnailing feature isn't yet implemented and I just missed it, feel free to close this one.

fr3ak2402, 2023-09-23:
The theme with the thumbnail of Creality K1 I am currently implementing...

closed 2023-09-23 by martinoshub